Roger-random/Sawppy_Rover #1: Enhancement - cherry pick dae  / collada for 3d printed parts to augment rover

johndpope, 2019-03-08:
https://bitbucket.org/theconstructcore/curiosity_mars_rover/commits/all

Screenshot here / maybe a basic shell could be propped over existing rover so it's more like curiosity. 
https://github.com/BujuNB/iOS-EV3-3D-simulation-on-Mars/issues/1

johndpope, 2019-03-08:
[image]

johndpope, 2019-03-08:
https://www.ebay.com.au/itm/1-10-RC-Car-Lexan-Clear-Body-Shell-200mm-Lexus-LFA-Will-Fit-Yokomo-TAMIYA-HPI-/121875609990

was imaging a cover like this - 
[image]

johndpope, 2019-03-08:
[image]

https://www.youtube.com/watch?v=CQc6gAWl4dY

johndpope, 2019-03-08:
https://www.youtube.com/watch?v=QuOAk0pgsGY

johndpope, 2019-03-10:
https://www.lego.com/en-us/service/buildinginstructions/search?initialsearch=42081#?text=42081
came across this volvo concept - looks mint

[image]

[image]

johndpope, 2019-03-10:
[image]
[image]
[image]

Roger-random, 2019-03-10:
A Lexan shell would be a good optional component for Sawppy builders who want to make their rovers resemble Curiosity not just in mechanical geometry but also in cosmetic appearance. At the moment my focus is not on the rover's looks, but if anyone else wants to make these cosmetic shells happen that would be fantastic!

Playville, 2019-03-10:
Hi Roger!
I just finished the mechanical build a few minutes ago and started planning on some “cosmetics”, especially on the body. As soon as I’m done id love to share the inventor-Files if you like to?

johndpope, 2019-03-11:
Hi @playville - there are some png images inside the official 3D project (first link at top) that I guess could be used as decals. 

Feel free to post screen shots of work in progress. Would love to see. When your happy you can submit a pull request to merge your updates into project. Welcome  to the world of open source.

johndpope, 2019-03-11:
https://bitbucket.org/theconstructcore/curiosity_mars_rover/src/94974385f14a1f229bf436f91552ab6384fd89b4/curiosity_mars_rover_description/meshes/tex_01.png?at=master&fileviewer=file-view-default

[image]

johndpope, 2019-03-16:
Found this - 

[image]

https://www.thingiverse.com/thing:3493597

[image]

https://www.thingiverse.com/thing:1606181

[image]

Roger-random, 2019-07-21:
I'm happy to share that one of the Sawppy builders has made great progress at building cosmetic add-ons to make Sawppy look like Curiosity Rover. Please see [Laura McKeegan's project CJ ("Curiosity, Jr.") rover on hackaday.io](https://hackaday.io/project/165094).

[image]

And for anyone else who wishes to undertake a similar project, I've shared all the [3D resources I've found for Curiosity rover](https://newscrewdriver.com/2019/06/25/curiosity-rover-3d-resources/). The best one seems to be the Blender model released by NASA JPL.

[image]

closed 2019-07-21 by Roger-random

johndpope, 2023-09-01:
[image]
[image]
[image]
![00003-630438868](https://github.com/Roger-random/Sawppy_Rover/assets/289994/6209cf4c-4f4c-4541-ad8
[image]
[image]
[image]
[image]
[image]
[image]
6-9f1c03370418)
[image]

hi @Roger-random  - you might be interested to play around with stable diffusion
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Here I just prompt
 "mars curiosity rover lexan body shell, technical drawing, pencil![Uploading 00
[image]
011-2492057212.png…]()

also not sure you saw this - 
https://www.youtube.com/watch?v=fY32WFY-LuM&ab_channel=RobotGarden