Roger-random/Sawppy_Rover #11: Things to maybe add to Parts List

adamgreen, 2019-07-10:
As I was reading through the documentation to put together a list of the things I have to order, I ended up adding a few things to my list that didn't show up in docs/Parts List.md

Maybe there should be a tools section to give a list of tools that someone who is newer to these type of builds (such as myself) might not already have in their possession?
* 8mm Reamer
* Metric Tap & Die Set (A M3 tap specifically)

In the hardware section, I don't see mention of any **M3x16mm socket head screws** but I do see 10 of them being mentioned in the differential assembly instructions. Should these be added to the main hardware list?

What about a list of electronic components required? No matter if a person uses an Arduino or a RaspberryPi, they will need the **LewanSoul BusLinker** hardware to interface with the LX-16A servos which are included in the list.

I just opened this Issue to start a conversation about this topic. I can draft up a PR in the future based on resulting feedback.

As this is my first issue here, I just wanted to thank you for all of the work involved in this project. I really appreciate it! There is an awesome amount of work here!

Roger-random, 2019-07-13:
Thanks for the feedback and opening this discussion! I always appreciate people who go through the instructions and send feedback based on their first hand experience.

Here's my understanding of what you've conveyed:

1. **M3x16mm socket head screws**: yes I agree they missing and should be listed on docs/Parts%20List.md. Related: Not only are they used on the differential, they are also used to install Sawppy wheels to their hub. I didn't create an entire page on wheel installation thinking it was straightforward. But that also meant I had just the line "Once all the wheels are bolted on to the suspension" without mentioning the M3x16mm fastener and associated washers. Oops.

2. **8mm reamer, M3 tap, etc.**: I don't think they belong on the parts list as they're not actually parts of the rover. However, I agree it would make sense to have a page dedicated to discussing tools involved in building a Sawppy. Outline all the goals and different ways to accomplish that goal. Like how the 8mm shafts would ideally be cut on a metalworking lathe, but in the absence of a lathe, they can be cut with a Dremel. As long as the goal is met.

3. **Electronics components**: I didn't place much emphasis on precisely outlining the electronics, because I anticipated people would customize and do their own thing anyway. Do you think it is valuable to have a single "prescribed" set of electronics directions? Side note: it is actually possible to make an Arduino communicate directly with the LX-16A servos in half-duplex, skipping the need for BusLinker, with a software serial port that transmits and receives on the same pin.

adamgreen, 2019-07-14:
@Roger-random Thanks for the response!

1. **M3x16mm socket head screws:** I think your response contains enough information for me to start creating a PR to address it.
2. **8mm reamer, M3 tap, etc.:** I could start a page for required tools: 8mm reamer, M3 tap, 1.5 and 2.5mm Allen key (hex key) wrenches. You make a good point about some of the other items like preparation of the 8mm shafts.
3. **Electronic components:** I agree with your comments on the electronics. I hadn't investigated the LX-16A servos in much detail yet. I was waiting for mine to arrive first. I had suspected that the BusLinker might not be required so I am happy to hear that is the case.

Roger-random, 2019-07-21:
Thanks for your additions to the documentation. (Sorry for the delay - I had missed my notifications.) I'll close this issue to reflect the merged changes.

You said you're waiting for your servos to arrive. Once you've had some hands-on time with them, I'm sure you'll have feedback on how we can improve Sawppy documentation for electronics components. 

Please feel free to open new issues when that comes up, I look forward to reading them.

closed 2019-07-21 by Roger-random

adamgreen, 2019-07-22:
> I'll close this issue to reflect the merged changes.

Cool! I had planned to do the same but you saved me the effort :)

> You said you're waiting for your servos to arrive. Once you've had some hands-on time with them, I'm sure you'll have feedback on how we can improve Sawppy documentation for electronics components.

Actually I was waiting for pretty much all of the bits and pieces for my Sawppy build to arrive. Most of the parts arrived last week (including the servos). 
[image]

I am just waiting on my 3D printer now. Once it arrives, I will start building. I have some software / hardware items that I want to dig into in the meantime. This includes hopefully playing with the servos this week.

adamgreen, 2019-08-23:
> Side note: it is actually possible to make an Arduino communicate directly with the LX-16A servos in half-duplex, skipping the need for BusLinker, with a software serial port that transmits and receives on the same pin.

I recently started playing with the LX-16A serial servos. Pretty cool! I actually ended up needing to use the half duplex method to communicate with the servos anyway as the Cortex-M3 microcontroller I was using couldn't sink enough current to drive that BusLinker board with its strong pull-ups. I took one of mine apart and posted a [few notes about what I found inside](https://github.com/adamgreen/Ferdinand20#lewansoul-lx-16a-servo-teardown).

Roger-random, 2019-08-25:
Thanks for the detailed teardown notes. This will be very useful guide when I take apart my LX-16A that [failed short at Maker Faire.](https://newscrewdriver.com/2019/06/09/sawppy-emergency-field-repair-at-maker-faire-bay-area-2019-steering-servo/)

adamgreen, 2019-08-25:
@Roger-random Are you hoping to repair your failed servo or just root cause it? Either way, good luck! Sounds like a fun debugging project.

Roger-random, 2019-08-26:
I don't expect to put that servo back into active use, I'm mostly curious to see if the cause of the failure could be determined. If it could be repaired and put into some secondary/backup role, it would be nice but not a goal.

I read through more of your Ferdinand20 project documents (beyond the LX-16A teardown) and it sounds like a great project! I worry about Sawppy's speed, though. I modeled after Curiosity which is no speed demon with a top speed of 4cm/sec. Sawppy's top speed is significantly slower than other RoboMagellan entries. And while you can put faster motors inside the wheels, the suspension geometry is also optimized for low speed operation and would not absorb the shocks of high speed impact with obstacles. Hopefully this is something you've taken into consideration.

adamgreen, 2019-08-26:
> I don't expect to put that servo back into active use, I'm mostly curious to see if the cause of the failure could be determined. If it could be repaired and put into some secondary/backup role, it would be nice but not a goal.

I saw [this thread on the Trossen Robotics forum](http://forums.trossenrobotics.com/showthread.php?14684) where a user discusses the analysis of a premature failure in one of their LX-16A servos. They indicated that the brushes on the motor had failed. From my teardown, I thought the servos looked like they were quite well built. However, the originator of that forum thread indicates that the motors are being provided with more than their rated voltage in the servos.

> I worry about Sawppy's speed, though.

I am aware of that issue. At this point, I care more about completing the course and not necessarily the speed at which it is completed. I am looking more for a platform to perform my electronic and software experiments. Going slower will make some of those experiments easier. I might switch out the drive motors in the future to make it a little bit faster. Even though it might not complete the course at the highest speed, the rover should still look pretty cool as it runs the course. Getting other people interested in robotics, especially young people, is another goal of running the rover in the competition. I also like the construction method you used to build Sawppy and the experience of building my own should help me with future robot designs of my own.