Roger-random/Sawppy_Rover #15: .prt files

Divepit, 2019-10-31:
Hey Sawppy team!

We're looking to build our own version of Sawppy but would like to replace the Aluminum profiles with cylindrical carbon fiber tubes. Would you be willing to share the .prt files with us so we could directly adjust them? Of course we'd be more than happy to share the outcome with you guys as well!

Cheers,
Marco and David

adamgreen, 2019-10-31:
Does the OnShape link at the bottom of [this page](https://github.com/Roger-random/Sawppy_Rover#other-project-resources) not give you CAD file access you need? I believe that there is still a free level for OnShape.

Roger-random, 2019-11-25:
Any update, @Divepit ? As @adamgreen stated, the original Sawppy CAD file is a Onshape document set to be publicly viewable by anyone.

https://cad.onshape.com/documents/43678ef564a43281c83e1aef/w/392bbf8745395bc24367a35c/e/4a687fb6a85c60c7ec293eb2

You are welcome to make a copy and customize for cylindrical carbon fiber tubes.

writemecas, 2020-02-13:
Hi there,

wanted to give an update for everyone trying to download the model. You can, you just need an Onshape account. There is an option to create a free (hobby/maker) account so no subscription needed. Then just select the tap you want, right click an download.

Cheers!

Roger-random, 2020-03-08:
I've confirmed the Onshape free plan is still available, just not prominently promoted which is not surprising for a product that doesn't generate revenue for the company.
https://www.onshape.com/products/free

https://newscrewdriver.com/2018/12/30/onshape-is-free-for-makers-but-theyre-less-eager-to-say-so-now/

closed 2020-03-08 by Roger-random