Roger-random/Sawppy_Rover #16: LX-16A Servo not available online from Amazon.

TheMightyCodeCoder, 2020-02-07:
Hello Sawppy_Rover team.  I ordered one LewanSoul LX-16A Servo on Amazon and received it.  However, I went back to order more and Amazon is out and says that it doesn't know when they will become available.  I am also having trouble finding the LewanSoul Bus Linker.  I am speculating that this is a Coronavirus thing that's affecting the distribution of these.  If I order straight from China the shipping is very expensive.  Anyone have any idea where I can get these pieces locally?

adamgreen, 2020-02-08:
I placed 2 orders for LX-16A servos this past summer. Between the two orders, they were temporarily out of stock for a bit so I had to wait. I will keep my fingers crossed for you that you have the same luck.

Roger-random, 2020-02-10:
It's quite possible the latest scare has disrupted flow of logistics, but I can tell you this has happened before even without a medical scare. Since the time I started this project I have noticed LX-16A and/or debug board occasionally running out of stock on Amazon with no estimated restock date. These periods of scarcity lasts a few weeks, roughly the time we'd expect a shipping container to travel across the Pacific.

If you have the luxury of a few weeks of time, you should be able to wait out the dry spell. If not, it's time to get creative to adapt other actuators to drive your Sawppy.

I am not aware of other LX-16A retailers within continental United States.

adamgreen, 2020-02-10:
When I just checked, there were 2 of the LX-16A with the not required brackets in stock:
https://www.amazon.com/LewanSoul-Real-Time-Feedback-Bearing-Brackets/dp/B0748BQ49M

adamgreen, 2020-03-04:
@TheMightyCodeCoder  I just checked and it looks like you can pre-order now for a March 12th in-stock date:
https://www.amazon.com/LewanSoul-Real-Time-Feedback-Position-Temperature/dp/B073WR3SK9

Roger-random, 2020-03-04:
And the asking price has gone up from $14.99/ea to $16.99/ea. Roughly what we expected to see for increased tariffs imposed by the current U.S. administration.

adamgreen, 2020-03-08:
Amazon says that they are now in stock.

Roger-random, 2020-03-08:
Amazon shows that for me as well! We have waited through this drought. Closing this issue until the next time...

closed 2020-03-08 by Roger-random