Roger-random/Sawppy_Rover #17: Parts list missing qty of 608 bearings

nbering, 2020-08-06:
I'm willing to do a PR for the addition, but while I was putting together my parts orders, I had trouble getting a count for the bearings.

If I'm not mistaken, the bearings qty breaks down as:

| Assembly | Assembly Qty. | Bearing Qty. | Subtotal |
| --- | --- | --- | --- |
| Steering Knuckle | 4 | 2 | 8 |
| Fixed Knuckle | 2 | 1 | 2 |
| Corner Steering Joints | 2 | 2 | 4 |
| Suspension Bogie Joints | 2 | 2 | 4 |
| Suspension Rocker Joints | 2 | 2 | 4 |
| **Grand Total** | | | 22 |

I'm not sure if my count is entirely accurate, because some of the part installation instructions are inconsistent as to whether the bearing count is for all instances of that assembly, or per-assembly. Steering knuckle in particular looks like it shows 2-per assembly but says "2 bearings". Steering knuckle lists 2 bearings as well, but it's hard to tell from the photos if that's 2 per-assembly or total.

nbering, 2020-08-06:
Don't know how I missed this. I've been over the docs a few times but never saw this master parts list.

https://github.com/Roger-random/Sawppy_Rover/blob/master/docs/Parts%20List.md#hardware

closed 2020-08-06 by nbering

Roger-random, 2020-08-09:
It's one thing to have information written down, but it's even harder to organize everything well so people who need the information can find them. Obviously I'm not there yet, but I'm glad you found it eventually.

nbering, 2020-08-09:
Just as feedback, the two places I would have found the qty breakdown helpful:
- The [Amazon list](https://github.com/Roger-random/Sawppy_Rover/blob/master/docs/AmazonAssociate.md)
- The [components list](https://github.com/Roger-random/Sawppy_Rover/tree/master/docs#mass-produced-components), where it describes the requirements for the 608 bearings