Roger-random/Sawppy_Rover #18: TTL Board Amazon Associates Link

ninjix, 2020-12-06:
The TTL Board Amazon Associates link no longer points to a controller. It now shows a LewanSoul LX-224 Serial Bus Servo.

adamgreen, 2020-12-07:
I can't find it on Amazon right now either. Sometimes they run out of stock of the LewanSoul parts and they take the listing down and then it pops back up again in a month or two when they get more in.

The servos can be controlled without this board. I have interfaced a mbed-LPC1768 microcontroller directly to the LX-16A servos on my Sawppy rover so I know it can be done. I don't know if an example of directly connecting them to an AVR based Arduino has been documented yet though.

Patronics, 2020-12-07:
Not an ideal solution, but it seems that the x5 pack option of LX-224 servos in that listing does come with one.

Roger-random, 2020-12-07:
Thank you for pointing this out. I can't find a replacement link either, so I'll leave it as-is until I find a replacement and hope it comes back after some time.

The sporadic availability is a recurring problem, so I'm planning to use something else for the next Sawppy version.

ninjix, 2020-12-07:
Thanks, Roger. Is there a `Develop` branch somewhere we can help you?

nbering, 2020-12-07:
I'd have loved to support you more by using the affiliate links where I could, but had similar issues. Availability in Canada is somewhat different from US Amazon.

For LX-16A specifically, I ended up ordering from AliExpress. Amazon listings with good prices seemed to have low, or no availability. And those with good availability had much higher prices than the $500 design goal would allow for.

adamgreen, 2020-12-08:
Looks like it might be available again under a new link:
https://www.amazon.com/Parts-Accessories-Linker-LX-16A-Controller/dp/B08B5T7JX5/

Roger-random, 2020-12-10:
Wow, they are asking $30 for what was formerly a $10 item. At that point it's tempting to pay $90 and get the LX-16A five-pack, which includes one of those boards along with five servos.

Playville, 2020-12-10:
Abou a year ago I ordered a bunch of LX-16a directly at Lewansoul. Having around 8-10 Motors left, so if anybody urgently need some of those, will give them to other “Sawppyians” for the price I paid once, was about 15€ I guess.

Roger-random, 2020-12-10:
Also: thank you @ninjix and @nbering for your willingness to help. Right now the most valuable contribution I can receive are information from Sawppy builders outside of the United States. I want to switch Sawppy V2 away from parts that have proven to be hard to get. I'm brainstorming alternatives and I would love to get your perspective.

I have activated GitHub's new beta "Discussions" feature, let's all take it for a test drive! Here's the link: #20

ninjix, 2021-01-04:
Looks like a couple TTL boards came back into stock.
https://www.amazon.com/gp/product/B073WRLJB2

Roger-random, 2021-01-10:
Confirmed - the $10 listing (not the scandalous $30 listing) is back in stock and online. All links from Sawppy documentation appear to be working again, both affiliated and non-affiliate links.

Thanks, everyone!

closed 2021-01-10 by Roger-random