Roger-random/Sawppy_Rover #27: Interest in supporting BLDC motors?

samuk, 2022-01-04:
Hi would you be interested in working on BLDC motor support? I'd personally be interested in the Ros2 version, but Arduino control may be possible too?

More on the idea here: https://github.com/mgonzs13/ros2_rover/issues/2

Roger-random, 2022-01-07:
I have no objection to BLDC motor support, but it won't be the highest priority for development. In order to verify and test the functionality, I'll have to buy BLDC motors and controllers myself. I can't justify that expense at the moment.

I followed your link and I see pointers to motors designed for camera gimbal mounts. Usually, such motors are not suited for driving vehicle wheels due to low torque. Do you plan to use a gearbox?

I am suspicious of the motor specifications listed. It says "230g" of torque, but grams are the wrong unit to express torque.

nbering, 2022-01-07:
I'm interested in at least giving it a try. I'm planning to order a few BLDC motors and something to control to do some controlled experimentation for my own experience.

My focus with BLDC research so far has been for field-oriented-control on actuators for an arm. Simply replacing the serial drive servos isn't something I'd investigated yet, but since a simple ESC reduces the complexity over FOC application boards, it might be more economical for wheel drive than what I'd been seeing for actuators.

Most controllers I'd seen were 1 or 2 channels, though. So that'd be my next concern after torque - that the electronics might get a little complex and/or specialized.

samuk, 2022-01-07:
@Roger-random Thanks, I'll poke at the idea a bit and see if we can get something working. I opened a discussion here https://github.com/samuk/Sawppy_Rover/discussions/3 if anyone wants to chat further on it.

@nbering one of the ideas with https://rosmo.io/ is that the driver boards are 'stackable' on the [M5stack profile](https://shop.m5stack.com/collections/stack-series) so we _hope_ we could stack 3x of the ESC boards on a single ESP32 for 6x6 control. This is all theoretical at this point. But we hope it's going to work! Ordering the first batch of boards some time next week.

closed 2022-01-07 by samuk