Roger-random/Sawppy_Rover #37: Clips (2n125 and 3n20)

lolia-olu, 2023-09-27:
Hi, 

We're using a Cura Lulz bot to 3D print the parts, however, for clips 2n125 and 3n20, we're experiencing some difficulty during printing as they are not printing smoothly. What extruder would you prefer we print with and is there anything else you think we should be aware of when printing the clips?

Thank you.

adamgreen, 2023-09-29:
I printed mine out of PETG on a Prusa MK3S, the same used for the rest of the parts. I did hit a few issues:
* I needed to print a large number of them at once so that each layer would take long enough to let the previous layer cool before starting the next one. Looks like I printed 20 of the 3n20 and 50 of the 2n125 ones according to the .3mf files I still have kicking around from those builds.
* I used a textured PEI surface for most of the other parts but the clips just didn't have enough bottom layer surface area to stick reliably to this surface so I switched and used my smooth PEI surface for the clips instead.

I still got some blobbing and a bit of warping but I still ended up with enough good ones to put together my printer. They don't need to be perfect to do the job needed and they can't really be seen anyway. Actually some of the oddness in the shape made them fit a bit more snug in the channel anyway :)

lolia-olu, 2023-10-05:
Thank you @adamgreen and @Roger-random. Very insightful and I will keep these tips in mind.

closed 2023-10-05 by lolia-olu