Roger-random/Sawppy_Rover #43: Print Wheel

Fibo27, 2024-12-31:
Hi, I am using a Creality slicer which is based on Cura. However, it is taking an inordinate amount of time to print the wheels (nearly 18 hours vs 7.5 as you mention). Can you please share the key slicer settings if you can?

Thanks

nbering, 2024-12-31:
The slicer settings and an overview of printing considerations is documented here:
https://github.com/Roger-random/Sawppy_Rover/blob/c9bb17d4c28b7518bbd493eec5f878033b617ee7/docs/Print%20Overview.md

One factor required to establish the print time for these settings was the print and travel speeds. But you could probably infer those from the specs for the printer that was used for the original project parts.

Roger-random, 2024-12-31:
Thank you, Nicholas! Here's a link that should go directly to the slicer setting section on that page you linked:
https://github.com/Roger-random/Sawppy_Rover/blob/main/docs/Print%20Overview.md#stl-slicer-settings

Good point I did not explicitly specify print and travel speeds. I had left those values in my slicer at their default values. Since I didn't look at them, I didn't think to write them down.

Also, a reminder this project is now five years old and I have not made significant updates. (It's been on my to-do list...) Slicer software and 3D printers have changed a lot in that time.

Thank you for your interest and I hope you find success.

Fibo27, 2024-12-31:
Thank you both. The Mars rover has been of an interest to me for a while. Great work done by you. 
So far I had been working on getting sensors and software running on a four wheeled robot. With advances in AI and ROS2 in particular, I plan to use Vision Language Model, Text to Speech besides the ROS2 framework. I will be using an Nvidia Jetson Orin given that Rpi and Arduino do not have the same capability. I have all of these working on a four wheeled robot but would like to incorporate it on the rover. CAD and mechanical design is an area that I need to work on more - again thanks for all the work. Wish you a great 2025

closed 2024-12-31 by Fibo27